dzurikmiroslav/esp32-evse #1: Could you tell me how to compile to ESP32?

adrian-zulnedi, 2021-05-14:
Nice to meet you . My name is Adrian Zulnedi . Thanks to make it Evse ESP32 , I really Like it.
But I don't know how can I compliing in my ESP32. Cloud you tell me to get detalis Informations?
Because it's doesn't have information details. Thank you for your reading.

adrian-zulnedi, 2021-05-14:
Using Arduino IDE But it'doesn't work. Think I must use Visual Studio or can  which filename (PATH) must to  compling in ESP32.

dzurikmiroslav, 2021-05-16:
Hi, you need ESP-IDF, in Eclipse with espressive plugin you can import as existing project. 
I realise that project needs more documentation... but first I want test it with real EV charing.

adrian-zulnedi, 2021-05-19:
thanks for your promptly reply. I hope your self made circuit can successful to EV Chargin, Because Next week I will chek it with my EV car. (I order that your self made circuit with parts )  
Another case
 I want to know added about 2 question. Could you teach me?
First Question
I need to know that your circuit board is available to using J1772?
Last Question
Your EsayEDA [esp32-devkit-evse]   have making [CURRENT SESSING] but I don't know    that functions. for what to use that Can you teach me?

Thank you for reading my issue.

https://oshwlab.com/dzurik.miroslav/esp32-devkit-evse
[image]

dzurikmiroslav, 2021-05-19:
Design is based on https://www.ti.com/lit/ug/tidub87/tidub87.pdf shuld support J1772 but I WARN YOU: NOT TESTED YET!

[CURRENT SENSING] is gpio for CT transformer, used for energy meter.

esp32-devkit-evse is designed only for one phase (ADC1_7 voltage sensing and ADC1_6 current sensing), 
You should see in https://github.com/dzurikmiroslav/esp32-evse/blob/master/cfg/esp32devkitc/board.cfg ENERGY_METER_L1_CUR_ADC_CHANNEL and ENERGY_METER_L1_VLT_ADC_CHANNEL values.

Energy meter can be disabled setting ENERGY_METER=none, session consumption will be calculated from max charging current.

adrian-zulnedi, 2021-05-25:
Thanks a lot.  I can compile your sourcecode. 
I think this source need wifi which can I rewrite that PATH about wifi name and password?

adrian-zulnedi, 2021-05-25:
Thanks a lot. Right now I can compile your source code .
But I don't know who can I using . I think need connect to wifi but I don't know which I must replace source. 
Sorry for Question many times. Cloud you teach me about this 
Constitution and which can I repalce the wifi ?

closed 2023-01-02 by dzurikmiroslav