dzurikmiroslav/esp32-evse #3: Hello you need Help to improve this project ?

brunohorta82, 2022-12-20:
Hello, I am working on one similar project, if you want we can't join energies and make a super evse based on ESP32 :).
Please let me know if you are interested.
my email: brunohorta82@gmail.com

dzurikmiroslav, 2022-12-21:
Hi, YES, it will be great, but unfortunately now this is one-man-show project :-) 
I need to write some documentation and guides...

vitalyxxl, 2023-01-21:
Problem with install project at my computer ......
( not found  files  at project dirs ) 
Please help me  continue !!!

dzurikmiroslav, 2023-01-21:
After checkout repository, did you fetch git submodules? 
`git submodule update --init`

vitalyxxl, 2023-01-22:
Hello friend ! You reply -  good !
Please , give me your email for best communication . If possible
Best regards !

vitalyxxl, 2023-01-22:
Installation does not go through GIT. An error is generated:
   error: invalid path 'components/peripherals/include/aux.h'  

I tried to install by downloading the ZIP archive and extracting it to the specified folder . 

My next step ?

dzurikmiroslav, 2023-01-23:
I suggest you to clone project by git:
*git clone https://github.com/dzurikmiroslav/esp32-evse.git
<https://github.com/dzurikmiroslav/esp32-evse.git>*
After clone, init git submodules:
*git submodule update --init*

Otherwise you must manually download and place submodules by .gitmodules
config file...

For build you need ESP-IDF v5.0 (
https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/get-started/index.html
)

dzurikmiroslav, 2023-01-24:
> Hello ! AGAIN ERROR while installation : wrong files : aux.c & aux.h
> ( about naming Wicrosoft information :
> https://learn.microsoft.com/en-us/windows/win32/fileio/naming-a-file )
> :( :(

Hi
I am developing this project on Ubuntu OS, I found out that "aux" file name
is not allowed on Windows :-)
I was renamed to "aux_io".

vitalyxxl, 2023-01-24:
Thanks for the correction ! 
Now the project install without errors and warnings.
Compilation and assembly   ( idf.py build)   goes without errors (there are a few warnings, but I think this is not critical .....)
Binary files are written    ( idf.py flash )      to the controller without any remarks. 100% ready.

When I reboot the board, I see the following entries in the monitor  ( idf.py monitor )  :

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clockdiv:2
load:0x3fff0030,len:7000
load:0x40078000,len:15356
ho 0 tail 12 room 4
**load:0x40080400,len:3824
0x40080400: _init at ??:?**

**entry 0x40080648**

-------------------------------------------------------------------------------------------------

Is this normal behavior?

No access point is found in the list of Wifi networks ..... I have no idea how to connect to the server EVSE :( ......
How can I proceed further to revive :) ?

dzurikmiroslav, 2023-01-25:
If that log was printed once It's good, if repeating continuously there is
some problem...
When you press button on gpio pin, who you defined in board.cfg
BUTTION_WIFI_GPIO
device should create WiFi AP with name esve-XXXXXX, after connect on it, go
to ip 192.168.4.1 there should be WebUI

vitalyxxl, 2023-01-26:
Thanks for the prompt response.
I made all the necessary settings related to the numbers of inputs and outputs ( LEDS/BUTTONS ...) for the ESP32  dev board.
There were no changes in the behavior of the board .... 
Experiments on other ESP32 dev boards  showed the same behavior. That is, there are no obvious signs that the loaded code is working correctly.
If anyone can give me a ready-made dump or files for controller firmware, 
I would be grateful. In general, has anyone managed to create a working project from this repository in the Windows 10 (64) environment. 
Maybe I'm not digging there? Although other projects are successfully assembled and quite efficient ....

Thanks to all who responded.

dzurikmiroslav, 2023-01-26:
Before compile & flash project, did you set right ESP-IDF configuration?
To deploy board config must be set `CONFIG_BOARD_CONFIG_DEPLOY=y` and `CONFIG_BOARD_CONFIG="custom"`.
While flashing, will be deployed board config from `cfg/custom/board.cfg`, you can edit it before flashing...
https://github.com/dzurikmiroslav/esp32-evse/wiki/Board-config

vitalyxxl, 2023-01-26:
Thanks for the clarification. But I made changes (just in case) to all three configuration files ..... It still didn’t work .....
But I'll check again. Apparently not today.

By the way, in my chargers based on I use 12V power at the input of the board (as in the original version). 
And I get negative  voltage (-12) volts using a specialized chip TC 1044 (TC 7660 ....).
I like this solution due to its low price ($0.3 -0.8 compared to the converter as in the original project), 
small dimensions and a minimum number of additional "strapping" components .... only two capacitors !

Why wouldn't the author of this project use this solution? 

Yes, there is a reworking of the scheme, but it's worth it! I'll take care of it anyway. 
The main thing is to somehow launch the project in its current form ......

Another question about the dimensions of the printed circuit board.
In my opinion, it is more logical to use the dimensions for installation in DIN RAIL cases . 
Then it will also be possible to place a small-sized power supply (and in the future, a LCD display (?) on the top cover) in one case. I'm sure this will be very true in terms of creating finished devices. 
It remains to add only a power contactor (relay) in common wall-box.

Do you think this proposal deserves consideration ? :)

PS: Is there another way to communicate with the author of the project? For example , email ? :)

dzurikmiroslav, 2023-01-26:
Many topics in last comment :) 
Please start separate discussions in https://github.com/dzurikmiroslav/esp32-evse/discussions
I'd like to talk about HW design with you...

closed 2024-07-26 by dzurikmiroslav