dzurikmiroslav/esp32-evse #24: ESP32-S3-N82 not working/not compiling

Alfred1975, 2023-10-23:
Hello!

I'm new in this field - ESP and so on. 

I bought ESP32-S3-N8P2 thinking that there will be no problem compiling for this chip. But nothing good happened. Its compiling, but after flashing chip rebooting with error "system_internal.c:158 esp-idf".

I google that a lot of people have problems compiling on s3 or moving from s2 to s3.

I tried a lot on menuconfig \ github and esp-idf forums etc but no chance to make it working.

Is it better to buy S2 as you use or maybe you know some hints why its not compiling?

Regards

jetpax, 2023-10-23:
This project certainly works for me on S3.

What do you have in your sdkconfig file re S3

eg I have

```
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0009
```

Alfred1975, 2023-10-24:
Dear Miroslav,

Project is compiling, in sdkconfig is the following:

CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_INIT_VERSION="5.2.0"
CONFIG_IDF_TARGET_ESP32S3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0009

But after flashing i see only cycling reboots with error "system_internal.c:158 esp-idf"

I tried different compile options that helped people from other projects that migrate from S2 to S3 but no goood news.

Do you have S3 to flash? Any ideas?

Regards

dzurikmiroslav, 2023-10-24:
Hi, unfortunately I don't have esp32s3 (I have only esp32 and esp32s2)

When you flasing are you deploying config partition (BOARD_CONFIG_DEPLOY) and which config are you using? (CONFIG_BOARD_CONFIG)

It may help to identify the problem to see the full log output...

Alfred1975, 2023-10-24:
Dear Miroslav,

From s2, renamed and added, so I have +1 folder in cfg named esp32s3 and there is a board.cfg from s2.

How to see a full log output? I do not have a JTAG....

Regards

Alfred1975, 2023-10-24:
I bought S3 because S2 at https://products.espressif.com/#/product-selector?language=en&names= and in official .pdf this module is marked NRND (NOT RECOMMENDED FOR NEW DESIGNS). Strange but true....

dzurikmiroslav, 2023-10-24:
Yes, I would have now bought the S3 too, but when I started designing the PCB, it didn't exist yet :-)

When you making custom board config directory you must also edit `Kconfig.projbuild` to make it work...

I suggest you to edit file `cfg/custom/board.cfg` and set `CONFIG_BOARD_CONFIG_DEPLOY=y` and `CONFIG_BOARD_CONFIG="custom"`

More info about config: https://github.com/dzurikmiroslav/esp32-evse/wiki/Board-config

Alfred1975, 2023-10-24:
I already done this. I changed the Kconfig.projbuild after a made cfg for s3 two weeks ago. And in menuconfig before flashing i selected esp32s3da

                choice BOARD_CONFIG
		prompt "Board config file"
		default BOARD_CONFIG_ESP32DEVKITC
		help
			Board config file
            
		config BOARD_CONFIG_ESP32DEVKITC
			bool "esp32devkitc"
		config BOARD_CONFIG_ESP32S2DA
			bool "esp32s2da"
		config BOARD_CONFIG_ESP32S3DA
			bool "esp32s3da"
		config BOARD_CONFIG_CUSTON
			bool "custom"
	endchoice

	config BOARD_CONFIG
		string
		default "esp32devkitc" if BOARD_CONFIG_ESP32DEVKITC
		default "esp32s2da" if BOARD_CONFIG_ESP32S2DA
		default "esp32s3da" if BOARD_CONFIG_ESP32S3DA
		default "custom" if BOARD_CONFIG_CUSTON

dzurikmiroslav, 2023-10-24:
OK, in `sdkconfig.default` are set `CONFIG_ESP_CONSOLE_NONE=y`, when you remove it from `sdkconfig` and set `CONFIG_ESP_CONSOLE_UART_NUM=0` you should see logs from startup like this:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7080
load:0x40078000,len:15584
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
I (31) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (31) boot: compile time Oct 24 2023 13:08:23
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 app0             OTA app          00 10 00010000 001d0000
I (91) boot:  3 app1             OTA app          00 11 001e0000 001d0000
I (98) boot:  4 cfg              Unknown data     01 82 003b0000 00004000
I (106) boot:  5 data             Unknown data     01 82 003b4000 0004c000
I (114) boot: End of partition table
I (118) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=58634h (362036) map
I (257) esp_image: segment 1: paddr=0006865c vaddr=3ff80000 size=00004h (     4) load
I (257) esp_image: segment 2: paddr=00068668 vaddr=3ffb0000 size=03dd0h ( 15824) load
I (269) esp_image: segment 3: paddr=0006c440 vaddr=40080000 size=03bd8h ( 15320) load
I (278) esp_image: segment 4: paddr=00070020 vaddr=400d0020 size=ddbe8h (908264) map
I (608) esp_image: segment 5: paddr=0014dc10 vaddr=40083bd8 size=13828h ( 79912) load
I (653) boot: Loaded app from partition at offset 0x10000
I (653) boot: Disabling RNG early entropy source...
I (664) cpu_start: Multicore app
I (665) cpu_start: Pro cpu up.
I (665) cpu_start: Starting app cpu, entry point is 0x40081534
0x40081534: call_start_cpu1 at /home/mdzurik/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (683) cpu_start: Pro cpu start user code
I (683) cpu_start: cpu freq: 160000000 Hz
I (683) cpu_start: Application information:
I (687) cpu_start: Project name:     esp32-evse
I (693) cpu_start: App version:      v0.2.0-4-gf2a730d
I (698) cpu_start: Compile time:     Oct 24 2023 13:08:21
I (704) cpu_start: ELF file SHA256:  707dc311454d980a...
I (710) cpu_start: ESP-IDF:          v5.1.1
I (715) cpu_start: Min chip rev:     v0.0
I (720) cpu_start: Max chip rev:     v3.99 
I (725) cpu_start: Chip rev:         v1.0
I (730) heap_init: Initializing. RAM available for dynamic allocation:
I (737) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (743) heap_init: At 3FFB8EA8 len 00027158 (156 KiB): DRAM
I (749) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (755) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (762) heap_init: At 40097400 len 00008C00 (35 KiB): IRAM
I (770) spi_flash: detected chip: generic
I (773) spi_flash: flash io: dio
I (778) app_start: Starting scheduler on CPU0
I (782) app_start: Starting scheduler on CPU1
I (782) main_task: Started on CPU0
I (792) main_task: Calling app_main()
I (792) app_main: Running partition: app0
I (912) app_main: Partition cfg size: total: 7781, used: 2259
I (912) app_main: Partition data size: total: 278861, used: 1255
...
```

Also I suggest you to start with minimal board.cfg like this, and gradually expanding it:
```
#Device name
DEVICE_NAME=ESP32 minimal EVSE
#Button
BUTTON_WIFI_GPIO=0
#Pilot  
PILOT_PWM_GPIO=33
PILOT_ADC_CHANNEL=7
PILOT_DOWN_THRESHOLD_12=2410
PILOT_DOWN_THRESHOLD_9=2104
PILOT_DOWN_THRESHOLD_6=1797
PILOT_DOWN_THRESHOLD_3=1491
PILOT_DOWN_THRESHOLD_N12=265
#AC relay
AC_RELAY_GPIO=32
```

Alfred1975, 2023-10-24:
Dear Miroslav,

The S3 board **is working** with your ESP32 minimal EVSE `board.cfg`

As you told me I will step by step expand to full version according to my soldering.

Thank you and regards!

closed 2023-10-30 by dzurikmiroslav