dzurikmiroslav/esp32-evse #27: Not compiling after last update #de2fd29

Alfred1975, 2023-11-13:
Hello!

I ran 'git pull' after last changes #de2fd29 and now i see the following:

In file included from D:/esp32-evse/components/script/src/be_mqtt_client_lib.c:8:
D:/esp32-evse/components/script/src/be_mqtt_client_lib.c: In function 'm_subscribe':
C:/esp/esp-idf/components/mqtt/esp-mqtt/include/mqtt_client.h:464:84: error: '_Generic' selector of type 'const char *' is not compatible with any association
  464 | #define esp_mqtt_client_subscribe(client_handle, topic_type, qos_or_size) _Generic((topic_type), \
      |                                                                                    ^
D:/esp32-evse/components/script/src/be_mqtt_client_lib.c:226:26: note: in expansion of macro 'esp_mqtt_client_subscribe'
  226 |             int sub_id = esp_mqtt_client_subscribe(ctx->client, topic, qos);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~
[1010/1062] Building C object esp-idf/script/CMakeFiles/__idf_script.dir/lib/berry/src/be_byteslib.c.obj
ninja: build stopped: subcommand failed.

I have the latest update of esp-idf v.5 and the rest I did not touch. 

What to update to make it compiling?

Regards,
Alfred

brunohorta82, 2023-11-21:
same problem

Alfred1975, 2023-11-21:
After changing to currently latest stable release is 5.1.1 of idf everything is OK

brunohorta82, 2023-11-21:
thanks i will try

dzurikmiroslav, 2023-11-21:
Described in #28

closed 2023-11-21 by dzurikmiroslav

brunohorta82, 2023-11-21:
solved with 5.1.1 but breaks with 5.1.2

dzurikmiroslav, 2023-11-21:
Fixed in commit cbc1e1c