dzurikmiroslav/esp32-evse #29: Charging scheduler

dialmak, 2023-11-21:
Hi Miroslav. 
Some countries have two-tariff meters. The price of electricity at night is significantly lower than during the day. I really wanted to add a real-time charging scheduler (of course, the time must first be set via the NTP server). You did a wonderful job, thank you.

dialmak, 2023-11-21:
For convenience, I added an RTC based on PCF8563 to the circuit.

dzurikmiroslav, 2023-11-21:
Hi, I also have two-tariff metering in mi garage :-)
My eletromer has HDO output (low tarif contactor) which is wired to digital input of evse, and this script reading it:
```
import evse
import aux

class hdo_driver
    def every_1s()
        var hdo = aux.read('IN1')
        if (hdo != evse.enabled())
            evse.set_enabled(hdo)
        end
    end
end

evse.add_driver(hdo_driver())
```

I plan extend scripting to add cron support like Tasmota https://tasmota.github.io/docs/Berry/#timers

dialmak, 2023-11-21:
The WEB interface is, in my opinion, nicer and more convenient.

dzurikmiroslav, 2023-12-18:
I was thinking about the web UI of scheduler, would setting the hour precision would be enough???
My camera has this setup: https://support.reolink.com/hc/en-us/articles/900000431446-How-to-Set-up-Continuous-Record-Schedule/#h_01GTDPY9FXPFC0PAAFRXW885YQ

Alfred1975, 2023-12-18:
Hi! 

I think hour precision is a good start :) because for meters its ok, for home charge its better to use scheduler and Amps reader (balancer) in junction box to give difference in Amps coming and free (for charging).

dzurikmiroslav, 2024-01-07:
I am working on scheduler... What should can be controlled by scheduler? Currently can control:
- enabled
- available
- limit charging current to 6A
- limit charging current to 8A
- limit charging current to 10A

More than one scheduler will be supported, what count of schedulers is useful?

dialmak, 2024-01-09:
Hi. In my opinion, there is no need to complicate the scheduler.
Something like - turn on the charger at some time, for example at 3 am. Limits current can be set globally in advance.
It’s more important to figure out how to simply (without entering the UI) enable charging at an unscheduled time if the scheduler is configured. Of course you can control it with a button...

dialmak, 2024-01-09:
As for the scheduler for days of the week or days of the month, I think there is no need for this. Enough time.

dzurikmiroslav, 2024-01-09:
The electricity supplier I was considering, for example, has discounted electricity on weekends or a tariff for heat pumps where even hours is cheaper than odd hours. Limiting charging current during certain hours could be useful in places, where other appliances than charging EV.

Yes, charging outside the scheduler will be possible. The schedule should only enable/disable charging at the falling/rising edge, so if charging is disabled by the scheduler, it can be enabled by a button and will be disabled on falling scheduler hour.

This is a draft version of a web interface:
[image]

dialmak, 2024-01-09:
Thanks.  It's fine.

dzurikmiroslav, 2024-01-10:
I working on it... in one week, maybe two ;-)

dialmak, 2024-01-10:
Wonderful.  You are the best :⁠-⁠)

dzurikmiroslav, 2024-01-17:
Scheduler is in master branch, to create first, click button Add scheduler.
I'm waiting for feedback ;-)
Also serial AT command mode was removed, check serial settings.