dzurikmiroslav/esp32-evse #30: Crash after Flash

brunohorta82, 2023-11-21:
Hello I get this loop crash and de device can't boot.

Test Board: ESP32S2 ESP32S3 ESP32-PICO-V4

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x17cc
load:0x4004b000,len:0x4
load:0x4004b004,len:0xaa0
load:0x4004f000,len:0x3128
entry 0x4004b1c0
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002fb74
0x4002fb74: _xt_context_save at /Users/brunohorta/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_context.S:126

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0x17cc
load:0x4004b000,len:0x4
load:0x4004b004,len:0xaa0
load:0x4004f000,len:0x3128
entry 0x4004b1c0

dzurikmiroslav, 2023-11-21:
I am suggest you to read issue #24 it was similar problem...

brunohorta82, 2023-11-21:
ESP System Settings
Channel for console output
Set to UART0 to solve the problem

Thanks

closed 2023-11-21 by brunohorta82