dzurikmiroslav/esp32-evse #35: How to Compile Project?

aapocan, 2023-12-15:
Hi everyone, thanks for project @dzurikmiroslav .
I have used only Arduino IDE before.
Now, i have eclipse with esp tools for this project. 
I did examples project blink etc. 
I open this project in eclipse and I couldn't do anything else.
thank you for your help

dzurikmiroslav, 2023-12-18:
In Eclipse, it is not enough to just open the project as a Makefile project, it is necessary to use the import an existing IDF project from the idf-eclipse-plugin.

I've been using VSCode with ESP-IDF for a long time now, I found Eclipse quite complicated, and I didn't enjoy reinstalling a configuring idf-eclipse-plugin after every new release ;-)

After checkouting code did you fetch GIT submodules? Please execute: `git submodule update --init --recursive`

aapocan, 2023-12-21:
thanks for answer. İ can not do it. Can you Check my dektop 
remote connection?

aapocan, 2024-01-12:
Thanks for your help. I think I managed to compile and install it. I haven't tried it on a car yet. The TL081CP seems to overheat when the circuit is idle, is this a problem?

dzurikmiroslav, 2024-01-14:
The operational amplifier should not produce much heat (no more than the ESP32). Do you have a genuine part? I managed to buy a bunch of fake parts on AliExpress :)

aapocan, 2024-02-10:
Yes I changed it and the problem was solved, thank you.
Now I'm trying it on the car, but it gives a warning waiting for authorization.

I click on the authorization button. but there is no connection
[image]

dzurikmiroslav, 2024-02-12:
It seems that the detects pilot state **B** even when no EV is connected... Which is voltage on CP? When has no EV connection (state **A**) it should be about 12V.

Problem should be caused by:
- wrong config of `PILOT_DOWN_THRESHOLD_X`
- bad power supply
- voltage drop on the operational amplifier
- wrong voltage divider in the CP circuit

I recommend you read the wiki about callibarion of CP https://github.com/dzurikmiroslav/esp32-evse/wiki/CP-calibration

closed 2024-07-26 by dzurikmiroslav