dzurikmiroslav/esp32-evse #44: Cp fault state e

demirs1, 2024-02-16:
I (119219) evse: Set max charging current 32A
I (119219) evse: Set charging current 76A*10
I (119219) evse: Set default charging current 100A*10
I (119229) evse: Set require auth 0
I (119229) evse: Set socket outlet 1
I (119229) evse: Set rcm 0
I (119229) evse: Set temperature threshold 60dg.C
I (120819) evse: Enter A state
I (120819) ac_relay: Set relay: 0
I (120819) pilot: Set level 1
I (120879) evse: Enter E state
I (120879) evse: Error bits 1
I (120879) ac_relay: Set relay: 0
I (120879) pilot: Set level 0

"I am getting 0V from the cp leg of the esp32-s2, it writes pilot error. When I check the circuit and measure the voltage, I see 1.45V at the tl081cp output. PILOT_PWM_GPIO=33 I get 0V PILOT_ADC_CHANNEL=3 I get 3.3V."

demirs1, 2024-02-16:
I am still seeing 1.45V at the cp_term output."

demirs1, 2024-02-18:
Why is the voltage output from the tl081cp opamp so low? Vout is 1.65V, there is no gain

dzurikmiroslav, 2024-02-19:
Was the operation amplifier bought from china (many fakes parts are on AliExpress)? I recommended you try buy operational amplifier from local shop. It doesn't have to be only TL081CP, it can be replaced with another that is suitable for the voltage and handle the frequency of 1kHz (for example LF353).
Have you connect on CP output TVS diode? try simulate without it.

demirs1, 2024-02-19:
I tried without TVS, but the result is the same. I think the op-amp I have might be fake. I will try with a new op-amp. Thank you.

demirs1, 2024-02-19:
"Opamp replaced and the issue is resolved. Thanks. Can you assist with another matter? Currently, I can perform the charging process, but when I try to start or stop the charging process on the interface, my vehicle encounters an error and doesn't stay in standby. It gives an error shortly after stopping the charging process, around 2 seconds. How can I put the vehicle into standby mode when there is no charging process? Is there a setting for this on the interface?"

dzurikmiroslav, 2024-02-20:
Enabled switch on main screen should be used to pause and resume charging process. Also in Settings / System / Charging available switch are set to EVSE to not available state (when is connected EV not socket lock will be acquired, session will be interrupted).
I tried to implement this scenario:
[image]
EN input is like Enabled switch in dashboard, C1/C2 is AC relay output
Source: https://www.phoenixcontact.com/en-il/products/ac-charging-controller-em-cp-pp-eth-2902802 in Downloads / Manual

If something doesn't fit according to this scenario, let me know.

If I may ask, what kind of EV are you trying it on? My EV also reports an error that it cannot charge, but after enabling charging, the charging process starts again.

closed 2024-07-26 by dzurikmiroslav