dzurikmiroslav/esp32-evse #46: OTA Update

Alfred1975, 2024-03-25:
Hello!

I tried to update from version 0.5.0 to 0.6.0 by air. I push the button (System::Firmware::Check for update) but nothing happened. I see red popup error "Cannot be fetch latest version info"

Error log is the following:
E (819032) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (819032) esp-tls-mbedtls: Failed to verify peer certificate!
E (819032) esp-tls: Failed to open new connection
E (819032) transport_base: Failed to open a new connection
E (819042) HTTP_CLIENT: Connection failed, sock < 0
E (819042) http_rest: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
W (819042) httpd_txrx: httpd_resp_send_err: 500 Internal Server Error - Cannot be fetch latest version info
W (819042) httpd_uri: httpd_uri: uri handler execution failed

Also I noticed that I receive 404 for  https://dzurikmiroslav.github.io/esp32-evse/firmware/ 

Or OTA feature is not working? Local update by bin file is working...

Regards

dzurikmiroslav, 2024-03-27:
Hi,
I don't know any other free hosting where I could distirbute a binaries for OTA. Currently I using for that github separate branch.
Problem is that github change SSL certificate (now has certificate from Sectigo CA, before has from DigiCert CA). In source code was bundled DigiCert CA root certificate, which is used during HTTPS OTA.

Alfred1975, 2024-03-27:
OK, have a good day!

dzurikmiroslav, 2024-03-28:
Fixed in 0.7.0, now using default certificate bundle.

closed 2024-03-28 by dzurikmiroslav