dzurikmiroslav/esp32-evse #49: CP not reaching -12V

rnunes22, 2024-04-09:
Hello, this project is amazing! I have tried multiple EVs, and it is working fine, but with a Skoda Plug-In, the car is throwing an error.  The EVSE is at -9V State B.

I used an oscilloscope and the TL081CP is not reaching the low supply voltage. Instead of being at -12V it is at around -10.7V. Everything else looks fine, the rise and fall time is under 2us and the PWM frequency is at 1kHz. Maybe it is because of the TL081 not being a rail-to-rail OpAmp (it is a genuine one).

I have read the EN61851-1:2011 standard and the only value that is not within the tolerances is the negative voltage. 

Any clue about what the problem may be?

dzurikmiroslav, 2024-04-10:
Hi,
I am study SAE J1772-2010, there is section about pilot line voltage ranges:
[image]

Earlier, when I connected the oscilloscope, I did not reach -11.4V but only -11.2V
[image]
I tested charging on three different EVs (Skoda, VW, BMW), so I didn't think much more about it...

I have a few questions:
1. Which negative voltage supply of TL081CP?
2. Do you have a TVS diode at CP out? If so, I would try to measure it without it.
3. Which board schematic do you have? ESP32DevkitC, ESP32 S2 DA or custom?

I definitely agree with that, a rail-to-rail OpAmp could improve the output CP voltage. But I couldn't find any in my local store...

rnunes22, 2024-04-10:
I am reading at the input supply a -12,5V on the TL081 (pin 4 VEE).
I tried reading with and without the TVS diode and the difference is negligible still under the +-0.6 tolerances and the car won't charge.
I used the schematic from ESP32 S2 DA.

I will try using a voltage comparator like the LM311 it should have a faster response time and the output should reach the supply voltage both on the negative and positive rails. I don't know any OpAmp that meets all the requirments of the protocol...

[image]

dzurikmiroslav, 2024-04-10:
Where are you measuring voltage? On CP or OpAmp pin (before or after 1k resistor) ?
When you charging EV which works charging, what is negative voltage?
When set EVSE to disabled state, CP should be -12V, what is voltage on CP without car.

demirs1, 2024-04-10:
[image]

I still couldn't configure the CP (Control Pilot) and my measurement values are like this, in real measurements they are like this.

rnunes22, 2024-04-26:
I have tried with a TL072 and now it is working. I used only 1 of the 2 OpAmp channels. The rise and fall time is bellow the 2us and the negative voltage is on the -12V rail.
[image]

dzurikmiroslav, 2024-04-27:
Good job, now changing Skoda plug-in?

rnunes22, 2024-05-06:
Yes! Thanks for your help and congratulations on this amazing project.

closed 2024-05-16 by dzurikmiroslav