dzurikmiroslav/esp32-evse #51: Flashing problem...

dariusz737, 2024-04-25:
Hello. Great project. But I have a problem to flash DevKIt from esp32s2-evse.bin. Everything seems to be ok, no error on flashing. But after I have such error in logs. Does it matter that my board is different from one on GitHub? 
[image]
[image]

dariusz737, 2024-06-16:
> Hi, are you sure you have ESP32-S2? That devkit looks like an ESP32...
> If you did not compile the project and only flash the compiled binary, did you also make the necessary partitions and config file?
> To begin, I would recommend building the project from sources and flashing it directly through the IDE.

Hello 

Is it possible to flash esp32-evse.bin with this tool - https://esp.huhn.me/  

Best regrads
Dariusz Kulik 

 <https://github.com/dzurikmiroslav/esp32-evse/releases/download/v0.7.0/esp32-evse.bin>Wiadomość napisana

dariusz737, 2024-07-05:
Hello 

Finały with help of my friend I have managed to flash ESP32s2 and even connect to Home Assistant. 
We are successful to read data, but failed to sent two basic commands, ON-OFF charging and limit charging current. 
Can you please help with this? 

Best regards
Dariusz Kulik

dzurikmiroslav, 2024-07-06:
Congrats to flash :-)
How do you read data from Home Assistant? HTTP API or Modbus RTU/TCP ?

dariusz737, 2024-07-06:
MQTTWiadomość napisana

Lagerkugel, 2024-07-14:
Hello Miroslav,

I am a Mac user and I am desperate about the Espressif IDE.

Would it be possible for you to write a short script for the initial compilation of the ESP32 code?

Or better:

Would it be possible for you to set up an esp32-evse.factory.bin to start it, which I could write to the ESP32 using any web installer and would immediately receive the required partitions and a generic configuration file. (See Tasmota)

I believe that many more users would realize the project if this point were checked off.
For me, the project would have died without your help.

cu Matthias

dzurikmiroslav, 2024-07-14:
Hi,
I am know esphome web installer and its good feature. Better write one HTML/js protect then shell script for every platform...
I will try find free space to implement it.

dzurikmiroslav, 2024-07-18:
There is ESP Web Tools installer like (Tasmota/esphome) https://dzurikmiroslav.github.io/esp32-evse/web-installer
In future I maybe implement Improv serial standard for setting WiFi connection...

closed 2024-07-18 by dzurikmiroslav