dzurikmiroslav/esp32-evse #67: Request for PWM setting to support of ISO 15118 to read car SOC

raspberrypi66, 2024-09-13:
ISO 15118 is a communication protocol standard specifically designed for the communication between electric vehicles (EVs) and charging stations, including AC chargers. Enables exchange of various data, including SOC.
To communicate with PEV EVSE need to change CP pwm to 5%

I hope you might add this feature to your project

dzurikmiroslav, 2024-09-13:
Yes, I studied ISO 15118 a bit. But it's not just about setting the 5% duty cycle :-D

Additional hardware is needed for power line communication (QCA7005), which on runs a TCP/IP stack...

raspberrypi66, 2024-09-19:
> Yes, I studied ISO 15118 a bit. But it's not just about setting the 5% duty cycle :-D
> 
> Additional hardware is needed for power line communication (QCA7005), which on runs a TCP/IP stack...

Yes but communication can be establish when PWM set to 5% duty cycle, people modify some of power line ethernet to communicate with PVSE

nagyrobi, 2025-05-25:
In theory a [TPlink TL-PA4010P v5.0](https://github.com/uhi22/pyPLC/blob/master/doc/hardware.md) could be used to access TCP/IP packets.