dzurikmiroslav/esp32-evse #71: Pilot fault after power outage

redonebebu, 2024-10-09:
Is it normal that after a power outage when the power comes back on again in I get a pilot fault and have to plug out the car first. I expected the car continued with charging. When I simulate this with resistors and diode I get the same result.

This is with the ESP32DevkitC EVSE and v0.10.1

dzurikmiroslav, 2024-10-10:
I would be interested to know exactly what kind of car it is? Yes, charging takes place in the C/D state, it cannot jump directly to the C state.
I have implemented this state diagram:
[image]

If the EV is connected before the EVSE is turned on, it should go to state B, not C.

redonebebu, 2024-10-10:
The car is a Tesla model Y but maybe I did not wait long enough. I just cut the power while charging and then turned it back on again and the error led immediately turned on. Maybe the car does a reset after some time. I will check it again later. 

When I connect the car with the Tesla Mobile connector and pull the plug out while charging and then plug it back in the car continues charging. When doing this with diode and resistance connected between PE and CP it goes back to charging state after 50 seconds.

The next time I test with ESP32-evse I will wait at least 5 minutes to see what is happening.

redonebebu, 2024-10-11:
Ok tested it again but the first time I also noticed the Amperes set by ESP32-EVSE did not corresponded with the vehicle. So I replaced the Chinese TL081 with one from a trusted shop and now everything is working correct. 

If I cycle the main power the car starts charging again immediatly. Lesson learned: don't trust Chinese parts.