dzurikmiroslav/esp32-evse #74: WEBAPI not working

taHC81, 2024-10-27:
Hi Miroslav, can't get API working... Here's an example request and reply. Any suggestions? Thanks! BTW awesome project!

```
PS C:\> Invoke-WebRequest -Uri "http://192.168.0.50/config/evse" -ContentType "application/json" -method GET | select *

Content           : <!DOCTYPE html>
                    <html lang="en" class="h-100">
                    <head>
                        <meta charset="UTF-8"/>
                        <meta name="viewport" content="width=device-width, initial-scale=1.0"/>
                        <meta name="color-scheme" content="light dark"/>
                        <link rel="shortcut icon" href="/assets/favicon-DjpvjAJf.ico"/>
                        <title>EVSE</title>
                      <script type="module" crossorigin src="/assets/index-BqOn_z--.js"></script>
                      <link rel="stylesheet" crossorigin href="/assets/index-ROnkTkQs.css">
                    </head>
                    <body class="d-flex flex-column h-100">
                    </body>
                    </html>

Encoding          : System.Text.UTF8Encoding+UTF8EncodingSealed
InputFields       : {}
Links             : {}
Images            : {}
BaseResponse      : StatusCode: 200, ReasonPhrase: 'OK', Version: 1.1, Content: System.Net.Http.DecompressionHandler+GZ
                    ipDecompressedContent, Headers:
                    {
                      Content-Type: text/html
                    }
Headers           : {[Content-Type, System.String[]]}
RawContent        : HTTP/1.1 200 OK
                    Content-Type: text/html

                    <!DOCTYPE html>
                    <html lang="en" class="h-100">
                    <head>
                        <meta charset="UTF-8"/>
                        <meta name="viewport" content="width=device-width, initial-scale=1.0"/>
                        <meta name="color-scheme" content="light dark"/>
                        <link rel="shortcut icon" href="/assets/favicon-DjpvjAJf.ico"/>
                        <title>EVSE</title>
                      <script type="module" crossorigin src="/assets/index-BqOn_z--.js"></script>
                      <link rel="stylesheet" crossorigin href="/assets/index-ROnkTkQs.css">
                    </head>
                    <body class="d-flex flex-column h-100">
                    </body>
                    </html>

RawContentLength  : 517
RawContentStream  : Microsoft.PowerShell.Commands.WebResponseContentMemoryStream
RelationLink      : {}
StatusCode        : 200
StatusDescription : OK
```

taHC81, 2024-10-27:
Similar for POST:

```
PS C:\> Invoke-WebRequest -Uri "http://192.168.0.50/config/evse" -ContentType "application/json" -method POST -Body "{ 'chargingCurrent':11 }"
Invoke-WebRequest: Specified method is invalid for this resource
```

dzurikmiroslav, 2024-10-27:
Hi thanks ;) REST API is on base path `/api/v1`, should try POST on `http://192.168.0.50/api/v1/config/evse`

taHC81, 2024-10-28:
I haven't found the base path mentioned anywhere, but indeed it works now, thanks for pointing me the right way!
[image]

closed 2024-10-28 by taHC81

dzurikmiroslav, 2024-10-28:
Yes, REST page in Wiki is outdated, I don't have much time to keep it actual...
Therefore, I recommend inspect the REST request/response through the developer tool in web browser in Web UI.