dzurikmiroslav/esp32-evse #92: Modbus garbage protection

nagyrobi, 2025-05-09:
I have a modbus master device which spits some garbage data on the serial port when boots, this triggers in esp32-evse a series of log entries:
```
W (1379427) serial_modbus: Invalid packet CRC
W (1379937) serial_modbus: Invalid packet CRC
W (1380437) serial_modbus: Invalid packet CRC
W (1380987) serial_modbus: Invalid packet CRC
W (1381507) serial_modbus: Invalid packet CRC
W (1382027) serial_modbus: Invalid packet CRC
W (1382557) serial_modbus: Invalid packet CRC
W (1383067) serial_modbus: Invalid packet CRC
```
Despite the fact that modbus commands sent after boot are valid, esp32-evse is not responding anymore. I can fix this two ways:
- I reboot esp32-evse from the web ui (System > Settings > Restart)
- I reinitialize the UART in esp32-evse from the web ui (System > Serial > Submit with no setting changed).

If I reinitialize it I get:
```
I (1540777) serial_modbus: Stopping
I (1540787) serial_modbus: Starting on uart 0
I (1540787) uart: queue free spaces: 20
```
And esp32-evse starts responding properly to modbus requests.

Apparently the initial garbage data breaks `serial_modbus` which errorneously keeps printing in the log "Invalid packet CRC" despite valid data incoming later.

nagyrobi, 2025-05-09:
And a few modbus questions:
- for register 101, EVSE error bits, can you please provide the table of what error corresponds to which bit?
- for registers 402, 403, I see the same temperature for both values. What does Low and High mean, what's the difference?
- for register 405, I only get "v1" for app version. Minor values missing, is this intentional?

dzurikmiroslav, 2025-05-12:
Error values is defined here: https://github.com/dzurikmiroslav/esp32-evse/blob/master/components/evse/include/evse.h#L11 

Temperature sensor is connected on 1wire bus, can be connected more than one, when is one these values are same.

Register 405 should be string, I must test this behavior...

nagyrobi, 2025-05-13:
> Temperature sensor is connected on 1wire bus, can be connected more than one, when is one these values are same.

Why you named Low and High?

dzurikmiroslav, 2025-05-13:
The most important value is High temperature.
For example, when you have 4 temperature sensors which measure values: 50, 45, 60, 55
High temperature = 60
Low temperature = 45

You can connect external sensors like this:
[image]

Should be attached to AC contractor and Type2 socket etc.

nagyrobi, 2025-05-13:
Wow. Nice.

dzurikmiroslav, 2025-05-30:
> And a few modbus questions:
>
> * for register 101, EVSE error bits, can you please provide the table of what error corresponds to which bit?
> * for registers 402, 403, I see the same temperature for [...]

I tested string registers (state, app_version) and looks good

[image]

dzurikmiroslav, 2025-05-30:
In version `1.2.0` were added modbus optimizations, now should not spamming _Invalid packet CRC_

closed 2025-05-30 by dzurikmiroslav