dzurikmiroslav/esp32-evse #97: Emergency Stop suggestion

nagyrobi, 2025-05-17:
I saw your suggestion for Emergency Stop using AUX IN, tested it.

I see it doesn't really behave like an emergency should, because it doesn't cut the power immediately. Analyzing the script it refreshes every 1s, that's still too long time for a real undesired emergency case.

My suggestion would be to wire a NC emergency button in series with the AC_RLY contactor, which would cut the power instantly. That would result in voltages dropping to 0 at the energy metering inputs - and that would be a contradictory state with what's expected in C and D, which could propagate to F. Logic propagation delay wouldn't be problematic in this case as the relay is already open, and that's most important.

The above could work only when in voltage+current mode. Unexpectef current drop wouldn't be suitable as the car could stop charging too, not safe to assume.

Other approach: 
To be sure that emergency button is the cause we could use a 2NO+2NC emergency stop button, where 1NC could be used in series with the relay, and 1NO could be connected to AUX. 
In the next revision of the PCB a "jumper" screw terminal could be added in series with Relay 1 coil, so we could add the emergency stop breaker button in series with Relay 1, not just the contactor.

dzurikmiroslav, 2025-05-19:
Yes, the hw wire connection is always more reliable than sw implementation.

By implementing the emergency button in Lua, I wanted to show the script's capabilities...

nagyrobi, 2025-05-30:
[This](https://github.com/dzurikmiroslav/esp32-evse/discussions/70#discussioncomment-10930745) example seems simpler, because it's just:

```lua
local aux = require("aux")

return {
  name = "Emergency stop",
  every1s = function ()
    local emstop = aux.read("IN2")
    if not emstop ~= evse.getavailable() then
      evse.setavailable(not emstop)
    end
  end
}
```
is it enough if I just put this `init.lua` wothout any `component.register`, if I don't want more?

closed 2025-05-30 by dzurikmiroslav