dzurikmiroslav/esp32-evse #102: WebUI About

nagyrobi, 2025-05-24:
In WebUI About screen I see these parameters listed incorrectly:

- Proximity pilot	Yes
- Socket lock	Yes
- Residual current monitor	Yes
- Three phase energy meter	Yes
- Energy meter	Current and voltage

Despite not having enabled the "Socket outlet", the "Residual current monitor" and "Three phases" options in the settings. I assume in this case they should be "No" on About page.  "Energy meter" lists "Current and voltage" despite being Dummy or Current sensing only

dzurikmiroslav, 2025-05-30:
The parameters in this table are not actual settings but HW capabilities.

closed 2025-05-30 by dzurikmiroslav

nagyrobi, 2025-05-31:
But how can these be set? For example
- i didn't install the pp components at all on the board
- no RCM module plugged in

Thus these are missing hw capabilities. In the settings, these corresponding checkboxes lead to think that the user adjusts the firmware to the hw capabilities, so it would be logic to reflect them in the about screen.

dzurikmiroslav, 2025-05-31:
It's configured thought board.yaml

nagyrobi, 2025-05-31:
Ok, so if I remove these, will it be correct?

```yaml
proximity:
  adcChannel: 2
  levels: [1650, 820, 430]

socketLock:
  aGpio: 20
  bGpio: 19
  detectionGpio: 34
  detectionDelay: 1000
  minBreakTime: 1000
```