dzurikmiroslav/esp32-evse #103: Temperature sensor stability

nagyrobi, 2025-05-30:
As long as the only DS18B20 sensor connected to the 1wire data bus is the one on the motherboard, it works fine. 

I connected a second one in parallel, with a 5m cable, placed in the Type2 connector case at the car end, to detect any possible temperature issues at the plug. (the 5x6mm power cable and a 4-wire flat data cable go together in a shrink tube from the EVSE enclosure to the Type2 plug). I found that occasionally it caused temp sensor fault error, which also triggered stop of charging.

To mitigate this, I placed a 3.6k pull-up resistor between DATA and 5V, working fine since then. My assumption is that the external DS18B20 sensor uses the 5V power source, while the onboard one uses the 3.3V, and the data line is pulled up against 3.3V, not 5V.

I know this is a bit risky due to ESP32S2 not being necessarily 5V tolerant on the GPIO pins - I guess it would be appropriate to also expose 3.3V to a screw terminal to properly connect the externally wired sensors at the same level.

Also reducing the pull-up to half would be beneficial to suppress noise on longer cables.

nagyrobi, 2025-06-01:
Assembled a second board, replaced the 12V connection on the screw terminal with 3.3V (scratching the path on the PCB and jumpering with a wire from nearby 3.3V bus); changed the pullup R3 to 2k2. Both sensors working now correctly.

gribmic, 2025-10-24:
Keďže garáž nemám, iba prístrešok pri poklese teploty ovzdušia pod 7st musím nastaviť nabíjací prúd aspoň na 10A ináč po cca 60s dôjde k prerušeniu nabíjania. Pri teplote okolo 0st ani 16A nestačí aby nedošlo k výpadku nabíjania.

nagyrobi, 2025-10-24:
Would be so much better if observations would be posted in English so more people could give a helpful hand to the problem.

gribmic, 2025-10-24:
You're right, I didn't think of that at all.

Since I don't have a garage, only a shelter, when the air temperature drops below 7 degrees Celsius, I have to set the charging current to at least 10A, otherwise after about 60 seconds the charging will be interrupted. At a temperature of around 0 degrees Celsius, even 16A is not enough to prevent the charging from failing.

p.s. I have a few PCBs left, I'm willing to give them away for shipping if anyone is interested.