dzurikmiroslav/esp32-evse #106: Version 1.2.0

nagyrobi, 2025-06-01:
Updated to version 1.2.0, here are a few observations:
- you forgot to update translation from https://github.com/dzurikmiroslav/esp32-evse/discussions/60#discussioncomment-13130169
- Nextion `loTemp`, `hiTemp`, `reboot` working OK.
- Nextion `avail` not working:

```
W (74807) serial_nextion: Unknown variable: avail
W (289217) serial_nextion: Unknown variable: avail
```

- is `avail` only a command, but not a variable (not in the variables table in the doc)?
- are `defConsumLim`, `defChTimeLim`, `defUPowerLim` only variables to read, but no commands to set?

dzurikmiroslav, 2025-08-28:
I could implement reading `sub avail` variable, but in next major release, not in minor or bug-fix. (I have revoked CI pipelines for support multiple channels OTA, and it's depends on some changes in board config).
Yes `defXXXLim` variables are only read only.

I suggest you for future serial communication AT commands https://github.com/dzurikmiroslav/esp32-evse/wiki/AT-commands I want it included in the next release.

nagyrobi, 2025-08-28:
For AT commands, only serial port is available? For testing, any Telnet-like network method?

dzurikmiroslav, 2025-08-28:
> For AT commands, only serial port is available? For testing, any Telnet-like network method?

No yet, maybe in future.

closed 2025-10-22 by nagyrobi