dzurikmiroslav/esp32-evse #124: Modbus - 317 register

Antrac1t, 2025-07-24:
Would you mind to check if 317 refister is read/write? From my test seems that its just read register, because i am not able to change status of it. 315 and 316 are ok, i can change them and also read them without any problem. Thank you

dzurikmiroslav, 2025-07-24:
Yes 317 should be R/W register.
When you change value in WebUI it will appear in modbus reading register?

Antrac1t, 2025-07-24:
yes, when i change value in Web then without any problem i will read changed value. But i am not able to change it

Antrac1t, 2025-07-24:
TCP

Antrac1t, 2025-07-24:
i think that we can close issue, because its caused that i am running on old version. Guess that from 1.2.0 was added 317 register.

dzurikmiroslav, 2025-07-24:
no, it`s my fault, I wrote always true if condition :)
```c
        if (value != 0 || value != 1) {
            return MODBUS_EX_ILLEGAL_DATA_VALUE;
        }
```

Antrac1t, 2025-07-24:
will you upload new release?

Antrac1t, 2025-07-25:
thx i will try it

closed 2025-07-25 by Antrac1t

nagyrobi, 2025-07-25:
> I create 1.2.2 version with fix

Does it contain the RCM fix too?

dzurikmiroslav, 2025-07-25:
> > I create 1.2.2 version with fix
> 
> Does it contain the RCM fix too?

No, sorry I forgot RCM... I just made 1.2.3